Duet3D/PanelDueFirmware #2: Splashscreen

vignesh141292, 2017-03-16:
Hi,

I like the latest firmware, it was easy to build and the STOP button was really useful.

I wanted to know if I can add a splashscreen with the company Logo? Can you guide me on the steps to add it.

Thanks

chandler767, 2017-11-13:
I created a version of this firmware for [Fusion3 F400 printers](https://www.fusion3design.com/) that implements a splash screen with a logo image and a progress bar. You can see how I did it here:
https://github.com/chandler767/FusionPanel/blob/4554400cc050552d439745b32a2cc29e31e9a395/src/PanelDue.cpp#L1212

jpatrick-09, 2017-11-25:
Chandler767: What tool are you using to convert the images into arrays?

chandler767, 2017-11-25:
@jpatrick-09 I recommend using [ImageMagick](https://www.imagemagick.org/script/index.php).

[Convert a PNG image to a C++ Array](https://www.imagemagick.org/discourse-server/viewtopic.php?t=30421)

dc42, 2017-11-30:
I'm using a patched version of bmp2c. I haven't had time do a proper fork of the project and publish it yet.

paboman, 2018-03-13:
How to make my own splash screen on boot ?

dc42, 2018-03-17:
Instructions are now included n the readme file.

closed 2018-03-17 by dc42