Duet3D/PanelDueFirmware #3: Post build step fail

bluesign2k, 2017-03-23:
Having followed the instructions to setup the build environment, the post build step to convert the elf into a bin fails due to build artefact having a fixed name eg. "PanelDue-7.0" rather than "${ProjName}-7.0". Changing the artefact name in the build settings to the latter resolves the issue.

dc42, 2017-03-23:
Thanks. The mistake I made was that in the build instructions, the project should be renamed to PanelDue instead of PanelDueFirmware. I have updated the build instructions, but I will probably update the post build step too in the next commit.

bluesign2k, 2017-12-17:
As of the current version I believe there is another error in the G++ Linker settings... I have to add quotes (") to the following for the -T and -Map switches: 
`-mthumb -static -mcpu=cortex-m3 --specs=nano.specs -Wl,--gc-sections -Wl,--fatal-warnings -T${workspace_loc:/${ProjName}/src/ASF/sam/utils/linker_scripts/sam3s/sam3s2/gcc/flash.ld} -Wl,--defsym,__stack_size__=0x1000 -Wl,-Map,${workspace_loc:/${ProjName}/${ConfigName}}/${BuildArtifactFileBaseName}.map`
to:
`-mthumb -static -mcpu=cortex-m3 --specs=nano.specs -Wl,--gc-sections -Wl,--fatal-warnings -T"${workspace_loc:/${ProjName}/src/ASF/sam/utils/linker_scripts/sam3s/sam3s2/gcc/flash.ld}" -Wl,--defsym,__stack_size__=0x1000 -Wl,-Map,"${workspace_loc:/${ProjName}/${ConfigName}}/${BuildArtifactFileBaseName}.map"`

dc42, 2017-12-19:
You will need to use quotes if the path to your workspace contains any space characters. Also I believe there is a bug in Eclipse that sometimes occurs when that form of path is used. I'll change it in the next release.

dc42, 2017-12-23:
Added quotes in latest commit.

closed 2017-12-23 by dc42