Duet3D/PanelDueFirmware #4: Won't flash

ccavanaugh, 2017-03-30:
I'm trying to flash the latest PanelDue-7.0.bin and get an error from bossac "file operation exceeds flash size"

It flashes correctly with  PanelDue-7.0-1.15c.bin

dc42, 2017-03-31:
Is this a binary that you built yourself? If so, how large is it? The maximum size is 128Kb for the standard version of PanelDue, although we have an OEM version with more flash memory.

ccavanaugh, 2017-03-31:
This was downloaded from Github.   I downloaded as a zip file just to be certain the binary wasn't munged up by the browser.   I has also tried using bossac on Windows and Linux and received the same error.

Github download is reporting a 131,176 byte file size or 128.1Kb assuming 1024 bytes per Kb

dc42, 2017-03-31:
The binaries you should be using are here https://github.com/dc42/PanelDueFirmware/tree/master/Release/Edge and the 7" version is 130728 bytes long.

ccavanaugh, 2017-03-31:
Will try with the Edge release...  I used the binary in the Stable release folder which is 131,176 bytes.

Thanks!

dc42, 2017-03-31:
Thanks, I've removed those over-sized binaries now.

closed 2017-04-03 by ccavanaugh