Duet3D/PanelDueFirmware #8: Fan Speed

IdefixRC, 2017-07-09:
Hi,

sorry if I missed this somewhere but which fan is controlled by the PanelDue touch option? 
I'm using it together with a Duetwifi and have my Extruder Temp controlled fans on ouput 0 and 1 and the cooling fan on port 2 but can't control the cooling fan via the panel.

Thanks !

IdefixRC, 2017-07-11:
please close the topic.....found out that the fan temp on the display controls Fan 1.
Would be nice to have settings to change this but for now I simply changed the fans physically over.

Btw, any plans to work on other GUI options? I have no idea how complex it is to create them in C++ but would be nice to have different options (e.g. larger buttons and configuration optimized screens - use the fulls pace also for 1 Extruder machines).

dc42, 2017-07-15:
The fan slider on PanelDue controls FAN0 by default, but if you have a tool selected and that tool defines a different print cooling fan, then it will control that fan.

We are somewhat memory-constrained on PanelDue, so we will need to change to a different processor before we can make substantial improvements.

closed 2017-07-15 by dc42