Duet3D/PanelDueFirmware #9: Changing idle heater active temp does not show until activated again

Alex9779, 2017-07-26:
When I change the active temperature of a currently inactive heater I see the new temperature for about 2 seconds after accepting, then the display reverts to the previous value.
But the value seems to be set because when the heater then gets active the new temperature is used and also displayed.

dc42, 2020-04-23:
Alex, did you close this because it is resolved now?

Alex9779, 2020-04-23:
Ummm no, actually I didnt test after I updated the panel a few weeks ago, I just cleaned my open and ancient issues...

Alex9779, 2020-04-23:
Ok tested it, actually it is not solved, important is that an other heater has to be active, then setting the temperatures of an inactive heater has that issue...

wilriker, 2020-09-29:
This should be fixed in 3.2-RC2. Can you retest with that one?

wilriker, 2020-09-29:
Please reopen if it is not working for you.

closed 2020-09-29 by wilriker