Duet3D/PanelDueFirmware #11: HowTo Upgrade & Flash Firmware

Kriechi, 2017-09-04:
I think this project could benefit from a README file and some upgrade instructions.
For future reference: after googling around, I ended up with:

Connecting the PanelDue via USB to my Mac.
Pressing the ERASE button to put the board into flashing mode (?).
Now it shows up as tty device.
Then I used the [bossac tool](https://github.com/UDOOboard/bossac/releases) to flash the new firmware with the firmware bin files from https://github.com/dc42/PanelDueFirmware/releases:
`./bossac-udoo --force_usb_port=true --erase --write --verify --boot PanelDue-v2-5.0.bin`

Final step, unplug USB and power cycle the PanelDue. Now the new firmware should be displayed in the *Setup* screen.

dintid, 2017-10-21:
Yes, a readme would be nice.

I wrote a blogpost recently on how to upgrade though: https://betrue3d.dk/paneldue-update-firmware/

dc42, 2017-10-22:
I have added a link to the instructions in the readme file.

closed 2017-10-22 by dc42