Duet3D/PanelDueFirmware #13: Changing extrusion multiplier for tools other than 0 not working.

dotorg, 2017-09-25:
Pretty much what the title says. I upgraded to the 1.17 release of the PanelDue firmware, and since doing so, changing the extrusion factor for anything other than the first extruder causes the extrusion factor for the first extruder to change. The values shown are correct, and the web UI works fine, so it looks like PanelDue is just simply sending the gcode to change it for the primary extruder, not the selected one. (And, not the active tool, either -- which tool is active has to bearing on it, as well, it just always sets it for the first extruder.)

dc42, 2017-10-06:
Thanks for your report. I confirm this is a bug.  PanelDue is sending parameter P for the extruder number, but it should be D. Will be fixed in next release.

dc42, 2018-06-13:
This was fixed.

closed 2018-06-13 by dc42