Duet3D/PanelDueFirmware #18: File listing capped at 2kb of entries

dotorg, 2018-02-21:
The list of files on the SD card seems to be limited to 2kb of total size (with nulls) by the Vector at FileManager.cpp:25

The end result is files returned by the Duet beyond 2kb don't show up in the UI. (From a practical standpoint, as it seems the list is ordered by date uploaded, it means newer files and directories just disappear once you hit 2kb)

If the firmware has enough available ram, boosting that size would probably help. Alternately, the code could pull the list from the controller on each page switch, and not hold the data from the other pages in RAM at all, I suppose. (Increasing the buffer size is only really kicking the can down the road, if RRF and PanelDue have differing limits on file name sizes/file counts.)

I'm happy to do a pull and do that, if it'd be helpful. Not sure when I'll have time to, but either are pretty easy changes, it looks like.

dc42, 2018-06-13:
I will increase this for the version 3 controller builds in the next release.

wilriker, 2020-10-13:
File list size has been increased to 4KiB for version 3 boards.

closed 2020-10-13 by wilriker