Duet3D/PanelDueFirmware #19: Post build steps not executing

jpatrick-09, 2018-02-26:
I've downloaded the latest files and when I build it within Eclipse, the project builds fine, and I am left with the .elf and .map files, however the .bin file is not being generated. I am using the 2017-q2-update of the arm toolkit. Do I need to change the post-build steps code for it to work with arm toolkit?  My current post build script is : arm-none-eabi-objcopy -O binary ${workspace_loc:/${ProjName}/${ConfigName}}/${BuildArtifactFileBaseName}.elf ${workspace_loc:/${ProjName}/${ConfigName}}/${BuildArtifactFileBaseName}.bin

Any help is appreciated!

dc42, 2018-02-26:
There appears to be an intermittent bug in Eclipse when paths of that form are expanded. Try changing the first parameter to:

${workspace_loc}:/${ProjName}/${ConfigName}/${BuildArtifactFileBaseName}.elf

and make a similar change to the second one.

jpatrick-09, 2018-02-28:
That didn't seem to work, I copied the code and made the same changes to both parameters, and got the same result. Any other ideas?

dc42, 2018-03-01:
What was displayed in the CDT build console?

jpatrick-09, 2018-03-09:
[Console output.txt](https://github.com/dc42/PanelDueFirmware/files/1797132/Console.output.txt)
Here is the console file. I don't see any reference to the post build script in the console at all. Thanks for your help.

jpatrick-09, 2018-03-16:
I figured it out! The issue was I had parallel build enabled in the C++ build settings. Once I disabled that, it generated the .bin as expected. Thanks for the assistance.

closed 2018-03-16 by jpatrick-09

dc42, 2018-03-16:
Interesting! I have parallel build enabled too and the postbuild step works for me. Are you any any chance using the internal builder?

jpatrick-09, 2018-03-17:
That is correct, I am currently using the internal builder. Is it more advantageous to use an external builder?

dc42, 2018-03-17:
I've no idea, I only use an external builder (make) because that's how my original system was set up.

Looks like it is a known bug, see https://bugs.eclipse.org/bugs/show_bug.cgi?id=199758.