Duet3D/PanelDueFirmware #20: PanelDue for Marlin

bwgeez02, 2018-03-04:
I was wondering if there would be any interest in merging some code with the beta Marlin 2.0.x. (beta for 32bit boards). Thanks!

dc42, 2018-04-23:
Hi, somehow I missed this when you posted it on 4 Mar. PanelDue talks GCode over a serial interface. So the only thing you would need to do in Marlin to support is is to support the GCodes it uses, which are the ones with JSON responses. Specifically, M408 S0 and S1, M20 S2, and M36. You don't need to support all the M408 response fields because PanelDue will use whatever it finds.

PanelDue already adjusts the GCodes it sends and what it displays when the host announces that it is Smoothieware or Repetier in the M408 response, to avoid using or displaying features not supported by those firmwares. We could do the same for Marlin if necessary.

bwgeez02, 2018-04-23:
Hey thanks! I was starting to think the Panel was going to be Duet only. I really think Marlin users would appreciate the option of having a large touchscreen. I know I would. 

The 2.0 Beta is coming along nicely, but I don't program; just test the firmware on my Due/Radds and provide feedback. I can add a feature request on the Marlin page if you don't mind. What would be the best way of going about it?

bwgeez02, 2018-05-03:
I submitted a feature request on the Marlin board. I'll see what they think about integrating the PanelDue.

bwgeez02, 2018-05-03:
Scott is already at it! 

https://github.com/MarlinFirmware/Marlin/issues/10609#issuecomment-386190326

dc42, 2018-06-12:
In PanelDue firmware 1.21 I added support to recognise host strings beginning with "Marlin". The action is the same as for Repetier, i.e. don't show standby temperatures, don't use G10 to set temperatures, don't quote filenames, and the gcode folder is the root of the SD card.

mfs12, 2021-05-10:
Refer as well to MarlinFirmware/Marlin/pull/10610

mfs12, 2021-05-10:
> In PanelDue firmware 1.21 I added support to recognise host strings beginning with "Marlin". The action is the same as for Repetier, i.e. don't show standby temperatures, don't use G10 to set temperatures, don't quote filenames, and the gcode folder is the root of the SD card.

Marlin firmware reports "Marlin" as firmwareName M408.

mfs12, 2021-05-10:
PanelDue is properly supported.

closed 2021-05-10 by mfs12