Duet3D/PanelDueFirmware #22: Macros buttons overlaying temperatures

timmit99, 2018-03-08:
I have run into a bug/issue with the latest PanelDue firmweare 1.20beta4 (File Name "PanelDue-v3-7.0.bin") 

"The first 4 macros are displayed on the Control page unless there are too many tools configured to leave room"

We are using the PanelDue on a machine where the first tool has 5 heaters, the second tool has one heater and the third tool has one heater. The macro buttons (as stated above) overlaps the temperature buttons for our other tools. The firmware seems to note that there are only 3 tools so it thinks it should be able to display the macros without overlapping anything. However, our 7 heaters are in the way and the buttons conflict. 

[image]

Also, because our tools have multiple heaters, the tool icon in the top row no longer correspond to the proper heaters below it. ( I.E. Tool2 icon is above a heater that is associated with Tool1) 
Below is a lovely representation of how the layout theoretically should.
[image]

timmit99, 2018-03-08:
Could the macro button display be a toggle-able setting in "Setup" to allow users to turn it on or off?

dc42, 2018-06-13:
Please re-test this using firmware 1.21.3.

timmit99, 2018-06-13:
Unfortunately the firmware does not fix this issue.
On the Print page it looks fine but the Control page still has this issue. This is the same as on the 1.20beta4 firmware running on our other screen

[image]
[image]

apphead, 2019-12-17:
Again, that is not even alpha release kind of issues. See https://github.com/dc42/PanelDueFirmware/issues/39#issuecomment-566737281

wilriker, 2020-09-29:
This has been fixed in 3.2-RC2.

wilriker, 2020-09-29:
Please reopen if it is not working for you.

closed 2020-09-29 by wilriker