Duet3D/PanelDueFirmware #24: Rotating Panel Display 90 Degrees

ajquick, 2018-03-15:
-label:enhancement

Are there any ways to rotate the display so that it is vertical instead of horizontal?

I have the 7" PanelDue, and the only way to secure it with my machine would be to put it vertically. Is there a setting to do it automatically? or would I need to generate my own release?

dc42, 2018-03-15:
The underlying drivers can handle portrait orientation, however the screen layout would have to be redesigned to fit all the buttons and data on the screen in portrait mode. the screen layout is defined by functions in file UserInterface.cpp and most of the constants for field sizes are defined in UserInterfaceConstants.hpp.

ajquick, 2018-03-16:
Would it treat the 7" portrait screen (480x800) like the 4.3" horizontal screen (480 x 272)?

Except with 2/3rds as wasted space?

ajquick, 2018-03-16:
Just adding a new bit of code in ```DisplaySize.hpp``` may do it?

`#elif DISPLAY_TYPE == DISPLAY_TYPE_ITDB02_70_R

# define DISPLAY_CONTROLLER		SSD1963_800
const DisplayOrientation DefaultDisplayOrientAdjust = static_cast<DisplayOrientation>(Default);
const DisplayOrientation DefaultTouchOrientAdjust = static_cast<DisplayOrientation>(SwapXY);
# define DISPLAY_X				(480)
# define DISPLAY_Y				(800)`

My experience with C++ is very limited.

dc42, 2018-03-16:
Yes, that's worth a try. I don't know whether the increased  DISPLAY_Y will cause any layout issues.

closed 2018-06-13 by dc42

drphil3d, 2020-04-02:
Any update on this? I Know there's a growing number of people that need this feature. 
https://forum.duet3d.com/topic/12647/another-stratasys-uprint-retrofit/17