Duet3D/PanelDueFirmware #25: Unable to print through PanelDue interface

coolstream1, 2018-03-21:
I have my 5-inch screen with the Panel Due 3.0 board up and running, but I cannot seem to get the Panel Due to start a print or run any macros. I can print fine from the web interface, but directly on the touchscreen, nothing seems to happen. The touch screen can however fully control the temperature, movement, and access the console fine.

In the console, it says GCode file ""1:/Cube.gcode"" not found Failed to open file "1:/Cube.gcode"
it does this for files on the SD card loaded onto the screens SD card reader, and for files loaded onto the duets card reader. But again, I can access and print those files fine from the web interface.

The M115 command returns: 
FIRMWARE_NAME: RepRapFirmware for Duet Ethernet FIRMWARE_VERSION: 1.19 ELECTRONICS: Duet Ethernet 1.0 FIRMWARE_DATE: 2017-08-14
I am using the PanelDue-v3-5.0 firmware version 1.2(15b1) and I am connecting the Panel Due to the Duet Ethernet with a ribbon cable.

dc42, 2018-03-21:
I think you are running the latest version of PanelDue firmware (1.20) but the version of main firmware you are running (1.19) is too old to handle quoted filenames. I suggest you update your Duet firmware.

coolstream1, 2018-03-22:
That was 100% it, thanks!

closed 2018-03-22 by coolstream1