Duet3D/PanelDueFirmware #29: Macro buttons on control page not working correctly when beeing in a macro sub folder

mloidl, 2018-06-07:
Hi,

When going into to macro subfolder the macro buttons on the control page are working.

I have the following directory structure inside 0:/macros/
```0:/macros/
0:/macros/BLTouch
0:/macros/commissioning
0:/macros/commissioning/Fil_Fan_off
0:/macros/commissioning/Fil_Fan_on
0:/macros/commissioning/Quadrat_Max
0:/macros/commissioning/XY_max
0:/macros/commissioning/XY_min
0:/macros/commissioning/Z-Cycle
0:/macros/filament
0:/macros/filament/Cleaning
0:/macros/filament/Color not clear
0:/macros/filament/Load NOW
0:/macros/filament/Load PETG
0:/macros/filament/Load PLA
0:/macros/filament/Unload
0:/stock/
0:/macros/1 Licht ein
0./macros/2 Licht aus
```

For the screenshot below i first switched into the folder "0:/macros/filament" and pressed the first button on the control page "1 Licht ein" two times
Then i switched into the folder "0:/macros/commissioning" and pressed the Button "1 Licht ein" on time

it seems that the macro name executed is not build correctly
[image]

Version: 1.21 (04b2)

dc42, 2018-06-07:
This sounds related to an issue that is already known and has been fixed in our latest internal build. Please test again when this build is released as firmware version 1.21 in the next few days.

mloidl, 2018-06-07:
I'll test with the next release.

Thank you.

mloidl, 2018-06-09:
Hi @dc42 

I've tested this issue with the new Release 1.21(08b2), but the Problem is still there.
Previously i'v build the last PanelDueFirmware-bins directly from source, now i wanted to check if this is maybe a problem of my builds.

I've tried to install the PanelDue-v2-4.3.bin Binary but unfortunatly the screen stays blank after update and reset. When i re-install my own build version (git tag 1.21) then it's working. I tried this multiple times.

dc42, 2018-06-09:
Is your PanelDue hardware definitely a version 2 or older?

mloidl, 2018-06-09:
I'm always installed v2-4.3.bin and also build it myself using this configuration. So i'm pretty sure i have a v2.

dc42, 2018-06-09:
Strange, because one of my own printers is using v2-4.3.bin so I know it works. Can you compare the downloaded pre-built binary with the one you built from source?

mloidl, 2018-06-09:
Filesize is different: The pre-built one is slightly larger

my build: PanelDueFirmware-v2-4.3.bin: 91820 bytes
pre-build: PanelDue-v2-4.3.bin: 92232

I'm using the build configuration as shown in the screenshot:
[image]

During writing this comment i tried again to send the pre-build binary to the PanelDue, but now it's working.

But i'm still able to reproduce the bug (also with the pre-build version).

dc42, 2018-06-12:
The original reported issue relating to macro buttons is fixed in release 1.21.2.

mloidl, 2018-06-12:
I will test it today and close the bug afterwards. Thank you very much.

mloidl, 2018-06-13:
Thank you. Was a pleasure to help.