Duet3D/PanelDueFirmware #37: I downloaded and added new language support, but now how do I turn the source code into *.bin?

crazyp1g, 2018-09-05:
I downloaded and added new language support, but now how do I turn the source code into  *.bin?

crazyp1g, 2018-09-06:
@gtjoseph   Thank you very much. The build is complete, but I still can't find where the *.bin file is.
make: *** [PaneIDue/src/ASF/sam/utils/cmsis/sam4s/source/templates/gcc/startup_sam4s.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [PaneIDue/src/ASF/sam/utils/cmsis/sam4s/source/templates/system_sam4s.o] Error 1
'Finished building: ../PaneIDue/src/ASF/sam/drivers/rstc/rstc.c'
' '
'Finished building: ../PaneIDue/src/ASF/sam/drivers/pio/pio_handler.c'
'Finished building: ../PaneIDue/src/ASF/sam/drivers/matrix/matrix.c'
' '
'Finished building: ../PaneIDue/src/ASF/sam/drivers/pwm/pwm.c'
' '
' '
'Finished building: ../PaneIDue/src/ASF/sam/drivers/pio/pio.c'
' '
'Finished building: ../PaneIDue/src/ASF/sam/drivers/pmc/pmc.c'
' '

03:45:27 Build Finished (took 3s.934ms)

gtjoseph, 2018-09-06:
Before building, you should have selected a build configuration that matches your panel.  For instance, I have the PanelDue 7i so I selected "Release-5.0i-7.0i".  After the build is complete, the f.bin file should be in the Release-5.0i-7.0i directory.

In your case...  you had an error building so even though you got the "Build Finished" message, that didn't mean that it was a successful build.  Look through the log to see what happened.

crazyp1g, 2018-09-06:
@gtjoseph    Thanks again!I forgot to select the build configuration, I chose the correct build configuration and now I'm prompted to save *.bin.

closed 2018-09-07 by crazyp1g