Duet3D/PanelDueFirmware #39: wrong layout when using only heater 2

nikscha, 2018-09-22:
the connector on the duet for heater 1 is broken, so i have to use the heater 2 connector on the duet while only using one heater in total. the webinterface can handle this no problem but the paneldue still shows the old heater. also the inputs seem switched on the panel due: when i enter a temp for heater 2, nothing happens. when i enter the temp over the webinterface, it shows up on the panel due. when i enter a temp on the panel due for heater 1 it shows up in  the field for heater 2 and everything works.

apphead, 2018-09-27:
I can confirm the behaviour outlined above. I have disabled H1 and H2 but they show up in PanelDue and the temp controls are misaligned by 2 columns. Working fine in DWC.
Please also note that macros NOT stored in a subfolder show up on the main screen and render the temp controls (for heaters 3-5 in the pic below) useless.
[image]

dc42, 2018-09-29:
This is a known issue. PanelDue is not currently aware of the mapping between heaters and tools. I plan to revamp how tools are displayed in PanelDue, but it will be some time before i can work on that.

apphead, 2019-12-17:
I have the same issue too. I know you are busy releasing Duet3, but all due respect, given the price tag of USD 100, that is a total deal breaker. Kind of sucks to find out that a otherwise top-notch setup is flawed when it comes to useability. Please fix this asap! Thanks

nikscha, 2020-04-03:
It's been about 1.5 years now since I opened this issue and almost a full year since the last firmware release for the panel due. Is someone working on this?

wilriker, 2020-09-29:
This has finally been fixed in 3.2-RC2.

wilriker, 2020-09-29:
Please reopen if it is not working for you.

closed 2020-09-29 by wilriker