Duet3D/PanelDueFirmware #46: Ability to set probe offset

mamoit, 2018-12-20:
Hello @dc42.
Would it be feasible to implement a probe Z offset calibration routine?
When I change the nozzle I have to re-calibrate the probe-nozzle offset.
The current method to set it through a `G31`, which requires access to the web UI or telnet, and therefore defeats the purpose on having a full fledged LCD.

mamoit, 2018-12-28:
I'm actually using a PanelDue, so I must be doing something wrong...
Can you set a Z offset to the current Z position and store it on the overrides?
How would you implement it purely with a macro?

wilriker, 2020-10-14:
You can manually send `G31` via the Console tab. I reckon this is not a very comfortable solution but at least one that would work.

mamoit, 2020-10-15:
@wilriker that's precisely my point when I stated the following
> The current method to set it through a G31, which requires access to the web UI or telnet, and therefore defeats the purpose on having a full fledged LCD.

wilriker, 2020-10-15:
@mamoit I don't think we are talking about the same thing. I meant to go to the Console tab on PanelDue, tap the keyboard icon and enter your `G31` command there. Your don't need DWC for that.

mamoit, 2020-10-15:
Ah, sorry, indeed I thought you meant "console tab" on the web interface.
I have not had a Duet since early 2019, so I'm not up to date with the features.
From what I recall the objective back then was to implement a button that an end user (without knowledge of GCODEs and inner workings of the machine) could press without having to dive into the console.

wilriker, 2020-10-22:
@mamoit Of course it is the primary intent of the PanelDue to abstract away deeper GCode knowledge. Though I think a problem here might rather be to find a one-size-fits-all strategy since probe offsets, bed levelling and everything else in this topic heavily depend on actual hardware setup.

I will close this GitHub issue now since this is the wrong platform for feature discussions. Please head over to the forum if you still think this is a topic to discuss as well as presenting what you actually had in mind this button is supposed to do. We can iterate, evaluate and develop this idea further then and eventually come back with a cleaner feature request here in GitHub.

closed 2020-10-22 by wilriker