Duet3D/PanelDueFirmware #50: work coordinates

drpeej, 2019-04-16:
Thanks for 1.23.2.   I'm really pleased to see the coordinates showing on the move screen

However,  would it be possible to show whether we are currently in the work coordinate or machine coordinate setting?

wilriker, 2020-09-29:
There is actually no way to be in the machine coordinate system for longer than one move. RepRapFirmware is always in one of the nine Workspaces (`G54..G59.3`).

I am planning on adding controls to store and modify these coordinates in version 3.3.

wilriker, 2021-05-21:
Current status: all relevant data is already fetched and maintained in the background. Still missing is a pop-up having the following elements:
* Selection buttons for WCS `G54..G59.3` that will let the user select which WCS to display data for. The current active WCS should be highlighted with e.g. having an asterisk appended
* For the selected WCS show the individual axis offsets
* Button to activate the currently displayed WCS iff it is not active
* Each axis offset value can be tapped to edit this value

mfs12, 2021-08-10:
@drpeej is this still relevant to you?