Duet3D/PanelDueFirmware #53: PanelDueFirmware development

saymtech, 2019-09-13:
Hello,
Sorry i am french...

I try to build the firmware with eclipse, and I have several errors.
You'll find log below.

It seems it's not find '_esplash' and other...

Could you help me ?
Thanks

Michaël

Building target: PanelDue-v2-7.0.elf
Invoking: GNU ARM Cross C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -O2  -g -Xlinker --gc-sections -Wl,-Map,"PanelDue-v2-7.0.map" -o "PanelDue-v2-7.0.elf"  ./src/Library/Misc.o  ./src/Icons/HomeIcons.o ./src/Icons/KeyIcons.o ./src/Icons/MiscIcons.o ./src/Icons/NozzleIcons.o  ./src/Hardware/Buzzer.o ./src/Hardware/FlashStorage.o ./src/Hardware/Mem.o ./src/Hardware/OneBitPort.o ./src/Hardware/SerialIo.o ./src/Hardware/SysTick.o ./src/Hardware/UTFT.o ./src/Hardware/UTouch.o  ./src/Fonts/glcd19x20.o ./src/Fonts/glcd19x21.o ./src/Fonts/glcd20x30.o ./src/Fonts/glcd22x32.o ./src/Fonts/glcd28x32.o  ./src/ASF/sam/utils/syscalls/gcc/syscalls.o  ./src/ASF/sam/utils/cmsis/sam3s/source/templates/gcc/startup_sam3s.o  ./src/ASF/sam/utils/cmsis/sam3s/source/templates/exceptions.o ./src/ASF/sam/utils/cmsis/sam3s/source/templates/system_sam3s.o  ./src/ASF/sam/services/flash_efc/flash_efc.o  ./src/ASF/sam/drivers/wdt/wdt.o  ./src/ASF/sam/drivers/uart/uart.o  ./src/ASF/sam/drivers/rstc/rstc.o  ./src/ASF/sam/drivers/pwm/pwm.o  ./src/ASF/sam/drivers/pmc/pmc.o  ./src/ASF/sam/drivers/pio/pio.o ./src/ASF/sam/drivers/pio/pio_handler.o  ./src/ASF/sam/drivers/matrix/matrix.o  ./src/ASF/sam/drivers/efc/efc.o  ./src/ASF/sam/drivers/chipid/chipid.o  ./src/ASF/common/utils/interrupt/interrupt_sam_nvic.o  ./src/ASF/common/services/delay/sam/cycle_counter.o  ./src/ASF/common/services/clock/sam3s/sysclk.o  ./src/ASF/common/boards/user_board/init.o  ./src/ColourSchemes.o ./src/Display.o ./src/FileManager.o ./src/MessageLog.o ./src/PanelDue.o ./src/Print.o ./src/RequestTimer.o ./src/UserInterface.o   
./src/PanelDue.o: In function `main':

D:\Documents\eclipse-workspace\PanelDue\Release-7.0/../src/PanelDue.cpp:1233: undefined reference to `_esplash'
./src/Hardware/FlashStorage.o: In function `FlashStorage::read(unsigned long, void*, unsigned long)':

D:\Documents\eclipse-workspace\PanelDue\Release-7.0/../src/Hardware/FlashStorage.cpp:49: undefined reference to `__flash_start__'
./src/Hardware/FlashStorage.o: In function `FlashStorage::write(unsigned long, void const*, unsigned long)':

D:\Documents\eclipse-workspace\PanelDue\Release-7.0/../src/Hardware/FlashStorage.cpp:101: undefined reference to `__flash_start__'
./src/ASF/sam/utils/syscalls/gcc/syscalls.o: In function `_sbrk':
D:\Documents\eclipse-workspace\PanelDue\Release-7.0/../src/ASF/sam/utils/syscalls/gcc/syscalls.c:90: undefined reference to `__ram_end__'

c:/program files (x86)/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/thumb/v7-m\libg.a(lib_a-writer.o): In function `_write_r':
writer.c:(.text._write_r+0x12): undefined reference to `_write'
c:/program files (x86)/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/lib/thumb/v7-m\libg.a(lib_a-readr.o): In function `_read_r':
readr.c:(.text._read_r+0x12): undefined reference to `_read'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [makefile:85: PanelDue-v2-7.0.elf] Error 1

dc42, 2019-09-13:
It looks like it has failed to pick up the linker input file, which is defined in the Misc options section of the linker tool settings. Try right-clicking on the project and select Refresh, then try building again.

saymtech, 2019-09-14:
Thanks dc42
In fact, i had problem with make installation, and  by dint of making changes.
I change the toolchain and parameters was lost especially those of linker.

After copying the file cproject again, it works,
although I still have error for uint8_t (and others) in paneldue.cpp

Thanks again

saymtech, 2019-09-17:
Hello I've another problem, i managed to compile , but after updating the firmware to the paneldue 7.0, the screen stay black. Is there something to modify in c++ for the type of panel ? or is there another thing i miss ?

Logs 

17:10:14 **** Incremental Build of configuration Release-7.0 for project PanelDue ****
make -j8 all 
Building file: ../src/Library/Misc.cpp
Building file: ../src/Icons/HomeIcons.cpp
Building file: ../src/Icons/KeyIcons.cpp
Building file: ../src/Icons/MiscIcons.cpp
Building file: ../src/Icons/NozzleIcons.cpp
Building file: ../src/Hardware/Buzzer.cpp
Invoking: Cross G++ Compiler
Invoking: Cross G++ Compiler
Invoking: Cross G++ Compiler
arm-none-eabi-g++ -DNDEBUG -D__SAM3S2B__ -DBOARD=USER_BOARD -DARM_MATH_CM3=true -DSCREEN_70 -I"D:\Documents\eclipse-workspace\PanelDue\src\config" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\thirdparty\CMSIS\Lib\GCC" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\cmsis\sam3s\include" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\utils" -I"D:\Documents\eclipse-workspace\PanelDue\src" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\cmsis\sam3s\source\templates" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\preprocessor" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\boards" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\header_files" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\boards\user_board" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\thirdparty\CMSIS\Include" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\pio" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\pmc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\services\clock" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\services\delay" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\wdt" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\pwm" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\uart" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\matrix" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\efc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\services\flash_efc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\rstc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\chipid" -Os -Wall -Wextra -c -std=gnu++11 -mthumb -MD -MP -mcpu=cortex-m3 -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -nostdlib --param max-inline-insns-single=500 -mlong-calls -MMD -MP -MF"src/Library/Misc.d" -MT"src/Library/Misc.o" -o "src/Library/Misc.o" "../src/Library/Misc.cpp"
Invoking: Cross G++ Compiler
arm-none-eabi-g++ -DNDEBUG -D__SAM3S2B__ -DBOARD=USER_BOARD -DARM_MATH_CM3=true -DSCREEN_70 -I"D:\Documents\eclipse-workspace\PanelDue\src\config" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\thirdparty\CMSIS\Lib\GCC" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\cmsis\sam3s\include" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\utils" -I"D:\Documents\eclipse-workspace\PanelDue\src" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\cmsis\sam3s\source\templates" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\preprocessor" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\boards" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\header_files" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\boards\user_board" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\thirdparty\CMSIS\Include" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\pio" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\pmc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\services\clock" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\services\delay" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\wdt" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\pwm" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\uart" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\matrix" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\efc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\services\flash_efc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\rstc" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\drivers\chipid" -Os -Wall -Wextra -c -std=gnu++11 -mthumb -MD -MP -mcpu=cortex-m3 -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions -nostdlib --param max-inline-insns-single=500 -mlong-calls -MMD -MP -MF"src/Icons/HomeIcons.d" -MT"src/Icons/HomeIcons.o" -o "src/Icons/HomeIcons.o" "../src/Icons/HomeIcons.cpp"
Invoking: Cross G++ Compiler
arm-none-eabi-g++ -DNDEBUG -D__SAM3S2B__ -DBOARD=USER_BOARD -DARM_MATH_CM3=true -DSCREEN_70 -I"D:\Documents\eclipse-workspace\PanelDue\src\config" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\thirdparty\CMSIS\Lib\GCC" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\cmsis\sam3s\include" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\common\utils" -I"D:\Documents\eclipse-workspace\PanelDue\src" -I"D:\Documents\eclipse-workspace\PanelDue\src\ASF\sam\utils\cmsis\sam3s\source\templates" -I"D:\Documents\eclipse-workspace\PanelDue\src [... cut ...]
Generating binary
arm-none-eabi-objcopy -O binary "D:\Documents\eclipse-workspace\PanelDue\Release-7.0/PanelDue-v2-7.0.elf" "D:\Documents\eclipse-workspace\PanelDue\Release-7.0/PanelDue-v2-7.0.bin" && cat "D:/Documents/eclipse-workspace/PanelDue/Release-7.0/PanelDue-v2-7.0.bin" "D:/Documents/eclipse-workspace/PanelDue/SplashScreens/SplashScreen-Duet3D-800x480.bin" >"D:/Documents/eclipse-workspace/PanelDue/Release-7.0/PanelDue-v2-7.0-logo.bin"
 

17:10:19 Build Finished. 0 errors, 28 warnings. (took 5s.412ms)

saymtech, 2019-09-23:
Sorry, i'll find it in configuration build.
I's OK now !!!

closed 2019-09-23 by saymtech