Duet3D/PanelDueFirmware #54: PanelDue 7i Firmware

Firedogs2x, 2019-09-26:
Hello, I have a PanelDue 7i Ver 2.0 that I purchased with my Duet3D Wifi board.
While moving it I pressed the rest button.
So I downloaded the latest version of BOSSA for windows 10.

No for my issue.
I tried multiple times to load firmware release 1.23.1 (both PanelDue-5.0i-7.0i.bin and nologo.bin) and I never was able to get the panel to start up.
Inside of BOSSA I selected "Erase All" and I even tried "Boot to flash" but nothing worked.

So finally I downloaded firmware release 1.22.1 PanelDue-5.0i-7.0i-nologo.bin and started up BOSSA selected the file and this time I noticed a difference in the list of selectable items. The "Brownout Detect" and "Brownout Reset" were no longer greyed out. So I selected "Erase All" and "Boot to flash" told BOSSA to "Write". Then I did a "Verify".
And now the panel is working.

I wanted to bring this to your attention for a couple reasons:
1) Uneducated users like myself need a little extra hand holding. So I know this is asking a lot but maybe a write up on how to use BOSSA specifically for this application would be nice.
2) Maybe there is an issue (incompatibility) with version 1.23.1 and that you are unaware of with some PanelDue screens.

Thank you for your support with this product.

Firedogs2x, 2019-09-26:
Sorry I forgot to mention that I used
https://betrue3d.dk/paneldue-update-firmware/
instructions on how to update

Firedogs2x, 2019-09-27:
Ok, everything worked just fine.

So I am sorry for the write up because it was my inability to read that caused my issues.

apphead, 2020-03-07:
@Firedogs2x  please close issue

closed 2020-04-08 by Firedogs2x