Duet3D/PanelDueFirmware #58: Changes to "Print" screen requests

JoshWobbles, 2019-12-01:
Have a few suggestions/requests;

- In the title bar (the first line on the screen), when printing, show the file name and possibly percentage complete
- I find the file and filament timers to be the least accurate when printing, can there be an option to use the slicer timer? It's what I primarily use when tracking jobs in the web interface.
- Change the "Print" page tab label to read "Info" or "Status" once printing has started

stevenwm, 2019-12-24:
Seconding showing the Slicer timer.  This is generally always within a few seconds of actual.  Maybe an option to swap the existing time left outs for just the slicer timer.

royal32, 2021-02-20:
Third request for adding slicer time. Filament/File/Layer time estimates are wildly inaccurate for me, while slicer time is accurate within 30 seconds.

wilriker, 2021-02-22:
Re time estimates please take your votes at
https://forum.duet3d.com/topic/21735/poll-time-estimates-on-paneldue-5-7
or
https://forum.duet3d.com/topic/21736/poll-time-estimates-on-paneldue-4-3

mfs12, 2021-05-10:
# priority from forum thread
simulated (if simulated) > slicer (if properly congigured) > filament > file > layer

# RRF calculated times

- layer time uses the average of previous layers and applies it to the remaining amount of layers.
- file time just looks at bytes processed vs. remaining bytes (no lines/instructions, just pure file 
- filament time is based on what the slicer puts into the file as amount of required filament vs. amount of filament already extruded.

The above three are all calculated within RRF and provided to PanelDue. Including 

# precaulated times

- simulated time
- slicer time

## comments from the forum thread
Main disadvantage is that they do no contain heating times so they will be very inaccurate for very short jobs but this will even out for longer jobs. Speed adjustments during printing don't update precalculated times. Also worth mentioning is that when DAA or more general forms of input shaping are used, the slicer time will be less accurate, because the slicer doesn't know what input shaping your machine is using. Simulated time should continue to be accurate. OTOH the slicer does know about things like the first layer being printed slower than the rest, and the last layers being printed slower if a minimum layer time is in effect. Filament-based time estimates take a while to catch up with print speed changes caused by those features.

wilriker, 2021-05-10:
The new remaining times have already been implemented in fcbc22a4504b3655d7ebcfd9e2450551a31852c6.

mfs12, 2021-05-12:
@wilriker can this issue then be closed?

wilriker, 2021-05-12:
I'd say it can. If the reporter is not happy they can reopen.

closed 2021-05-12 by wilriker