Duet3D/PanelDueFirmware #64: PanelDue with Radds & Repetier

Leon751, 2020-04-14:
Hi there, I bought a PanelDue 7 and I want to use it with the Radds 1.5 and Repetier firmware. I connected it to the serial port and installed the current firmware 1.23.2 on the PanelDue. I have connections with the Radds, can control the printer (move, home, heat). The speed is too fast when move, the speed is correct when home. I think the printer runs at max. Speed ​​that is stored in the Repetier firmware instead of the defined speed for move. Can you change it? Reprap firmware works with mm / min and Repetier with mm / sec. maybe that's where the mistake lies. The other problem is the SD card. I can access the SD card in the Radds, but I cannot start programs from it. The PanelDue jumps back to the start menu as soon as I start a program. After that, the printer can no longer be controlled; you have to switch it off and on again. The SD card slot on the PanelDue shows me no files from the SD card. I do not know if it is defective or the firmware from the PanelDue cannot. Is it possible to configure the firmware for the PanelDue so that it runs well with the Repetier firmware and programs can be played from the SD card slot on the PanelDue? Excuse my English, I'm using a translator.

mfs12, 2021-05-21:
# issue 1
- connected PanelDue with Radds 1.5 and Repetier firmware via serial port using paneldue 1.23.2 firmware
- connection works and can control radds printer
- speed is too fast, but right when "homing"

# issue 2
- sd card can be accessed using radds
- when program is started paneldue goes back to start menu
- after that printer cannot be controlled anymore, printer needs to be reset then

# question
- can paneldue be properly configured to run with repetier firmware?

dc42, 2024-12-09:
Operation with Repetier firmware would only be possible if Repetier was extended to support M409 and return at least those parts of the RRF Object Model that PanelDue uses.

closed 2024-12-09 by dc42