Duet3D/PanelDueFirmware #68: Feature request: Repeat last print

JostSchenck, 2020-05-03:
While printing a lot of face shields for a local institution I would have been really grateful if there was just a button "repeat print", selecting the last print, so I don't have to go back through sd card menu and try to get the right file with my clumsy fingers etc. I think printing the same object many times is (especially these days) a common task that would justify a shortcut. 

I am still on the very old 1.15c, because I never had a reason to update. I scanned through the changelogs and got the impressions something like this isn't implemented yet.

wilriker, 2020-10-13:
You are correct in your assumption that this has not been implemented.

Are you still interested in this feature? I would add it to the milestone for version 3.3 then.

wilriker, 2020-11-18:
This has been implemented and will be part of the next release after 3.2.3.

closed 2020-11-18 by wilriker

mfs12, 2021-06-23:
Wrong milestone.