Duet3D/PanelDueFirmware #71: Feature request: Touch lock

boa-pe, 2020-05-26:
It would be nice to have a lock function. This would prevent triggering unwanted actions by accidental display touch.

Just a button that needs to be long pressed to lock/unlock touch functions.

wize-wiz, 2020-12-04:
Absolutely a must have.

mfs12, 2021-05-10:
# cosiderations
That would be good. I quite frequently hit the STOP button by mistake. Though perhaps when locked, STOP should still be accessible, for emergencies!

mfs12, 2021-08-09:
Related to #169.

dc42, 2025-05-09:
We don't plan to do this. PanelDue is being superseded by a new screen. I will continue to implement changes where they are needed to retain compatibility with new RepRapFirmware versions for some time, but it is unlikely that we will implement new features.

closed 2025-05-09 by dc42