Duet3D/PanelDueFirmware #72: Not able to generate .bin files

STEPS-09, 2020-06-30:
Hello,
[image]

I just wanted to modify the code a little bit and see what kind of customization i can make to the PanelDue screen. I have a 7 inch display and i was able to upload the already existing .bin files which i downloaded from this forum.
I wanted to make some changes so i followed the instruction in the file Buildinstructions.md. I have installed eclipse and also gcc-2017-q2 update the compiler generates .elf file but i am not able to get .bin files. 

I don't have make.exe so I am using Internal builder.

How do i generate .bin files ?  The screenshot is below. 
If make.exe is compulsary please do guide as to how to add make.exe.

fluxfocus, 2020-07-11:
Hi STEPS-09,

-------TLDR--------
What I did was to pass the command to the os command shell rather than get eclipse to try to execute it itself. the objcopy tool location would have to be in the shell $PATH.
Again, i'm running on debian linux, if your os has some other command shell you'll have to use some other invocation.

**The original command is:**
arm-none-eabi-objcopy -O binary "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.elf" "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" && cat "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" "${ProjDirPath}/SplashScreens/SplashScreen-Duet3D-800x480.bin" >"${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.bin"

**You could replace it with:**
/bin/sh -c ' arm-none-eabi-objcopy -O binary "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.elf" "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" && cat "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" "${ProjDirPath}/SplashScreens/SplashScreen-Duet3D-800x480.bin" >"${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.bin"'

---TLDR---

From the little that I can see in the console readout, it looks like the arm obj-copy tool failed.
I have the external builder running and tried the internal and had a look at the difference in the console output when the command is executed.
When the external builder runs its script, each path is surrounded with the double quotes "....path.../blah/blah" and the binaries are created correctly.
When the internal builder runs it's post build step command, the double quotes are missing and the command fails.
I'm running this on debian linux. I'd be curious to know if it is an issue in windows.

I'm led to believe that the way eclipse executes this command, it is not done in a shell. I guess whatever the command mangler is, it struggles to deal with the double quotes.

We can instead ask eclipse to run the whole command in a shell.
To find the command, in eclipse go to the Project Properties. Then C/C++ Build -> Settings.  Scroll across the tabs (probably default shows tool settings) over to Build Steps. There is an area on that tab called Post-Build Steps. The objcopy command is there in the command box.

What I did was to pass the command to the os command shell rather than get eclipse to try to execute it itself. the objcopy tool location would have to be in the shell $PATH.
Again, i'm running on debian linux, if your os has some other command shell you'll have to use some other invocation.

**The original command is:**
arm-none-eabi-objcopy -O binary "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.elf" "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" && cat "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" "${ProjDirPath}/SplashScreens/SplashScreen-Duet3D-800x480.bin" >"${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.bin"

**You could replace it with:**
/bin/sh -c ' arm-none-eabi-objcopy -O binary "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.elf" "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" && cat "${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}-nologo.bin" "${ProjDirPath}/SplashScreens/SplashScreen-Duet3D-800x480.bin" >"${ProjDirPath}/${ConfigName}/${BuildArtifactFileBaseName}.bin"'

fluxfocus, 2020-07-11:
Sorry, i should say that I was using the 5 inch display project to experiment with this.
If the post-build command is different for the 7 inch or whatever active configuration you started with, just add the /bin/sh -c and the single quote marks around the whole original command.

dc42, 2020-07-13:
Since build variable ArmGccPath already needs to be set, I suggest replacing arm-none-eabi-objcopy by ${ArmGccPath}/arm-none-eabi-objcopy instead.

dc42, 2020-07-13:
I've made that change to the .cproject fie.

closed 2020-07-13 by dc42