Duet3D/PanelDueFirmware #74: 3.2RC1: no macros displayed, almost

joaobarros, 2020-09-18:
So Macros won't show any files, most of the times.
I was able to see my Macros again but cannot reproduce how I did it.
Adding and canceling a job and rebooting the Duet2 Wifi worked *1* time for each action but mostly macros do not show

joaobarros, 2020-09-21:
Issue still present in 3.2RC2

wilriker, 2020-09-29:
Sorry, I missed to get back to you earlier.

It works on my setup with Duet 2 Wifi and Duet 3.
In this case I need your `config.g` and a list of macro names to reproduce it.

Also please state your RRF version as well as if you mean the macros pop-up, the small list of macros on the Control tab or both.

joaobarros, 2020-09-29:
Hi,

I'm running: RepRapFirmware for Duet 2 WiFi/Ethernet version 3.2-beta1 running on Duet WiFi 1.02 or later
I'm using: PanelDue 7" v3.0

Macros don't appear in both macros pop-up and small list of macros on the Control tab

[paneldue_config.g.zip](https://github.com/Duet3D/PanelDueFirmware/files/5302413/paneldue_config.g.zip)
Attached: all my macros on the root folder:
[macros.zip](https://github.com/Duet3D/PanelDueFirmware/files/5302412/macros.zip)
All the macros I have on a folder named "Tunes":
[Tunes.zip](https://github.com/Duet3D/PanelDueFirmware/files/5302411/Tunes.zip)

Let me know if you need additional info.
Thanks!

wilriker, 2020-09-30:
Thanks for reporting. I have found and fixed the issue in 3.2-RC3.

closed 2020-09-30 by wilriker

joaobarros, 2020-10-01:
I can confirm that the issue is resolved.
Thank you!