Duet3D/PanelDueFirmware #80: Parsing error in `M409 K"<key>"` response leads to endlessly fetching this request again

wilriker, 2020-10-15:
Currently a `M409 K"<key>"` request is considered done when it was successfully parsed. Failing to do so will lead to repeating this request. Valid cases of this behavior are if RRF ran out of Output Buffers and responded with the appropriate error code.
If instead parsing fails because e.g. the JSON structure is valid but too large this will lead to an endless loop of requesting this data and failing to parse it.

To prevent this problem a counter should be implemented to repeat the same request a maximum of three tries, log an error if it still fails then and move on.

wilriker, 2021-05-10:
Code to handle parser errors is already prepared in `PanelDue.cpp` `void ParserErrorEncountered(const char*, const char*, const size_t[])`

As of right now this method does only log a parser problem to the Console tab. It needs extension to count how often the same request failed (based on `currentResponseType`) in  row and after a threshold (e.g. 3 times) it has to block this type of request and inform the user. A user then would have to open an issue in the forum and provide configuration so it can be debugged why the same response failed multiple times in a row.

In the past this was only seen once where the JSON path was too long to fit into `String<100> fieldId;` in `SerialIo.cpp`.

mfs12, 2021-05-10:
> If instead parsing fails because e.g. the JSON structure is valid but too large this will lead to an endless loop of requesting this data and failing to parse it.
@wilriker but if the full structure doesn't fit in the buffer, then it can't be a valid JSON structure.Or am i missing something here?

wilriker, 2021-05-11:
I think we are talking about different things:
if RRF runs out of buffers it will reply with `{err:-1}`, that is already covered and will result in poll-rate slow-down if this happens too often in 10s.
On the other hand, imagine a valid JSON structure that results in a key like
```
result.move.axis^.compensation.[lots of more nested objects]
```
This can eventually exceed the 100 character length that is currently reserved for `fieldId`. (In the past there was a case were the previous size of 50 was exceeded and thus I increased it to 100). Right now I _think_ there is no OM path that generates a `fieldId` exceeding 100 characters but it might somewhen in the future. This should though not block the user from using PanelDue at all (because it will never get past `Initialization`) but only render some functions not available and inform the user so they can inform us.

Of course in some rare cases a hickup in data transfer can turn a otherwise valid JSON into garbage. Therefore the method should count that this happens multiple times in a row to be certain it is not noise on the line.

PR #108 (2021-05-20): Improve seqs requests with errors
Closes #80
PR #112 (2021-05-20): Improve seqs requests with errors
Closes #80.