Duet3D/PanelDueFirmware #81: [Feature Request] Select Wifi network SSID & enter password though PanelDue

T3P3, 2020-10-20:
Related request:
https://github.com/Duet3D/RepRapFirmware/issues/441

Forum Post:

https://forum.duet3d.com/topic/18993/show-and-connect-to-available-ssids-through-paneldue

wilriker, 2020-10-20:
I can look into adding this to a milestone once the necessary changes in RRF have been implemented.

dc42, 2021-04-28:
I have added a F1 option to the M587 command in RRF 3.3RC1 to return the SSID list in JSON format. However, in firmware 3.4 we may add the SSIDs to the object model anyway, in which case we should remove the new M587 option to save code space. So best defer implementing this feature until we have decided whether to add the SSIDs to the OM.

mfs12, 2021-04-29:
suggested format of returned wifi network list.

```
{
    "rememberedNetworks": [
        {
            "ssid": "alice-wifi",
            "ip": "192.168.1.150",
            "gw": "0.0.0.0",
            "mask": "0.0.0.0"
        },
        {
            "ssid": "bob-wifi",
            "ip": "192.168.1.150",
            "gw": "0.0.0.0",
            "mask": "0.0.0.0"
        },
        {
            "ssid": "carol-wifi",
            "ip": "192.168.178.250",
            "gw": "0.0.0.0",
            "mask": "0.0.0.0"
        },
    ],
    "err": 0
}
```

if there is an error retrieving details, then you should get this:
```
{"rememberedNetworks": [],"err":"some error reason text"}
```

mfs12, 2021-05-11:
@dc42 do you recommend to wait for the updated object model? Or would it make sense to already implement this feature now and switch later to the object model? The major part is probably to build the UI and doing the request.

What do you think?

wilriker, 2021-05-11:
Just to state that here again, the initial request was less about switching between known networks but about showing the network scan results, selecting a new network from the list and enter the password. As of now there is no information coming back from Wifi module to RRF to publish the scan results. So this needs even more than extending the OM.

But yes, building the UI, hidden behind `#if` could be started already.

progzmaster, 2022-12-27:
Even if the current RRF does not support returning the visible Wifi networks, thus selecting from a list is not feasible, a simple window with two editbox for the SSID and Password would still be a very convenient addition.

The functionality could just construct a proper M587 command from those, maybe issue the sometimes necessary M552 commands before/after, possibly wait for the connection to succeed before losing the entered data, in case there is a typo.

I have just one printer and by the time I have to move it to a different network, I usually already have forgotten everything about the M587 command and have to google it every time. Apart from that, the editbox in the console is short, so typing in an M587 with long SSID and password is a pain anyway.

dc42, 2022-12-29:
RRF 3.5 together with the new DuetWiFiServer supports a new function to scan for visible wifi networks. So it will be possible for PanelDue to provide a list that the user can select from.