Duet3D/PanelDueFirmware #82: Modernisation with LVGL

JPluess, 2020-10-25:
Hi there

I stumbled  upon LVGL...

LVGL is an open-source graphics library providing everything you need to create embedded GUI with easy-to-use graphical elements, beautiful visual effects and low memory footprint.  

Lvgl.io

Would be nice to see some light animations...

Thanks

wilriker, 2020-10-26:
Thanks for bringing this to attention. I will have a look at that though I personally would rather not do any animations since they take time and a snappy (but nicer looking) interface take precedence.

mfs12, 2021-05-03:
# LVGL 

- https://lvgl.io/features
- 16,32 or 64 bit microcontroller or processor
- 16 MHz speed is recommended
- Flash/ROM: > 64 kB (180 kB is recommended)
- RAM: 8 kB (24 kB is recommended)
- 1 Frame buffer: in MCU, external RAM, or display controller

mfs12, 2021-05-03:
Version 3 along with 5i v1 and 7i v1

ATSAM4S4B processor, 120MHz, 64kB RAM, 256kb flash.
Version 2 manufactured from August 2016

ATSAM3S4B processor (64MHz, 48kb RAM, 256kb flash).
Version 1.0, 1.1 and earlier 2

ATSAM3S2B processor (64MHz, 32kb RAM, 128kB flash)

dc42, 2022-11-21:
The current PanelDue hardware has insufficient memory to run LVGL, however we plan to use LVGL or another framework when we migrate PanelDue to more powerful hardware.

T3P3, 2023-08-31:
A switch to LVGL would be a different hardware and screen firmware, if done.

closed 2023-08-31 by T3P3