Duet3D/PanelDueFirmware #85: Any guidence on adding Chinese language?

RayforLoy, 2020-12-01:
There are 20000 Chinese charactors, which are too much to add. About 4000 of them are commonly used, but are discrete in UTF-8, but GLCD Font Creator only support contionous charactor.

wilriker, 2020-12-02:
I'm sorry to say that I see no good solution here. 4000 characters is already a lot and a huge amount of data by itself. The current font only supports under 400 characters. I am not certain if 4000 extra characters would even fit into flash memory.

The only way I see is to first generate all 20000 continuous characters and then thin out to the 4000 required ones. But that might be very tedious if it cannot be automated.

dc42, 2020-12-02:
To add some detail, each supported character needs a 1-byte column count, followed by 2 bytes per column. So a character of width 10 pixels excluding spacing needs 21 bytes. Multiply by 20000 and you get about 420kb. The SAM4S4B used in PanelDue has only 256kb flash memory in total.

If you supported just 4000 characters, they would take up about 84kb, which would fit although you would probably need to remove the splash screen. You would also need a translation table to translate the Unicode point into the index in the table.

You could replace the ATSAM4S4BA-AU chip by ATSAM4S16BA-AU which has 1Mb flash memory, then you could probably accommodate all 20K characters.

wilriker, 2021-04-29:
Please start an topic in the forum if you need further assistance.

closed 2021-04-29 by wilriker