Duet3D/PanelDueFirmware #88: Small Color issue!

JPluess, 2021-01-08:
Hi there,
when I connect only Duet2Wifi v1.04 with PanelDue v3. (no sensors connected)
I get this strange look.
Thanks.

[image]

wilriker, 2021-01-08:
This usually is a result of flashing the wrong version (display drivers differ slightly between the various versions). Can you please state
* PanelDue hardware version
* PanelDueFirmware version flashed

JPluess, 2021-01-08:
Hi there, 
Firmware are the latest stable. 3.2.7. before i had 3.2.6 with the same issue.
it's is only with the light theme heatbed button, dark 1 and dark 2 seem to be ok. 
i installed PanelDueFirmware-v3-5.0-nologo.zip!

dc42, 2021-01-08:
You haven't told us what PanelDue hardware you have. If it is a PanelDue 5i then you have flashed the wrong firmware binary.

JPluess, 2021-01-08:
Dear, 
thanks, it was the correct binary. :) i have now flashed the wrong file 5i, and now the colors are different, instead of red its blue! But the issue with the heatbead having 2 colors remains.   

[image]

JPluess, 2021-01-08:
Nice! I can provide a video!
when loading, it first does the correct beige color. 
Than it overlays green and purple!

https://user-images.githubusercontent.com/7995240/104055057-672cfd00-51ee-11eb-96ce-d0e22eb3e0cf.mp4

I hope I could answer your question. 
PanelDueFirmware-5.0i-7.0i.zip -> being the wrong file for my board, makes a color shift to blue. heatbed color issue is there.
PanelDueFirmware-v3-5.0.zip -> displays correct color, heatbed color issue is there.

wilriker, 2021-01-08:
I cannot reproduce this on a 5i and there is no code that does paint buttons in two different colors.

A couple of ideas and questions:
* What happens if your switch tabs to either Console or Setup and then back? Does this fix the rendering for this button?
* If you tap that button will this change anything?
* Can you try rotating the screen 180° in Setup and see if this happens still?
* From the color it paints I guess your heater is in fault state, correct? If not which state is it in? Does changing the state e.g. via DWC fix the rendering?
* You said you have this issue in this 3.2.7 and also in 3.2.6. This implies you had a working version before. Please state which one that was.

dc42, 2021-01-08:
There is a colour gradient displayed on the Setup page. What does it look like?

JPluess, 2021-01-08:
Hi there, 
- Release 1.24 is working correctly!
- Release 3.2 is faulty, 

- Visiting another tab and coming back shows the same dual color icon.
- Pressing the icon, makes it filled with the same color as the notification screen
[image]

-  Rotating does not help.
- Here is the color gradient:
[image]

i have tested it on 3 different screens. All have the very same Issue.
The display with a 40Pin connector and a separate screen driver board V3! 

as soon as it wants to load the temp, it "breaks" something.

JPluess, 2021-01-08:
- Resetting the heater fault with M562 P0 produces this result:
[image]

tximy, 2021-01-09:
Put the correct firmware for your board (C firm over V3)

wilriker, 2021-01-09:
With a heater in fault state I can reproduce the behavior. Thanks for reporting. I will investigate the issue.

JPluess, 2021-01-09:
Perfect! I am glad we found the issue.
Have a good day😀

wilriker, 2021-01-11:
This will be fixed in the next release.

commit 52662357 (2021-01-11): Fix color gradient for heater-fault on light theme (fixes #88)

Restore color gradient on dark theme 1
closed 2021-01-11 by wilriker