Duet3D/PanelDueFirmware #92: PanelDue 3.2.8 Issues

dang472, 2021-01-19:
1) Tool icon blinks red for no reason when no tools selected
2) Even though part fan is 100% in web interface. The paneldue shows 0.

wilriker, 2021-01-19:
Hi @dang472 part 1 is already fixed in 3.2.9.
Re fan: PanelDueFirmware shows the fan speed of the currently selected tool. If there is no tool selected it will not show fan speed. But I will change that in the next version to default to fan 0 then.

closed 2021-01-19 by wilriker

dang472, 2021-01-19:
Hi, For part 2. The printer was printing with a tool selected and still had the fan speed at 0. I have the default fan 0 mapped to all my tools. I installed 3.2.9 and I noticed part 1 was fixed.

wilriker, 2021-01-19:
I just tried and I cannot reproduce the issue with the fan not showing the correct value if a tool is selected (if none is selected it will show the initial or previous tool's value - which might be zero).

Please test again and if the problem persists provide your `config.g` as well as the output of
```
M409 K"fans" F"d99v"
M409 K"tools" F"d99v"
```

dang472, 2021-01-19:
When I installed 3.2.9 it seems to be fixed. If it manifests again I will let you know. While printing the sliced gcode file increased the fan speed. The web interface was correct. But the paneldue showed zero.