Duet3D/PanelDueFirmware #93: PanelDue 7I colour issue

RootCNC, 2021-01-19:
Hi everyone,

I've recently purchased a Duet 3 6HC and a PanelDue 7i from Ooznest (Literally two days old) and been updating the files etc of the Rapsberry Pi, Duet and the LCD

However, updating to the latest version Release 3.2.9 I get these results: please help.
[image]

I've attempted to use load the following with the same effect:

- PanelDueFirmware-v3-7.0-nologo.zip
- PanelDueFirmware-v3-7.0.zip
- PanelDueFirmware-v3-7.0C-nologo.zip

I did double check and the IC is the ATSAM4S4B ie V3 from the files list :
[image]

The original version that was shipped with the device was 1.23.2 - if that makes any odds 

Any thought?

beikeland, 2021-01-19:
use the 5i-7i binary

RootCNC, 2021-01-19:
Wheeeeyyyyyy Back in business! thank you :)

Apparently I'm as blind as a bat! thank you

closed 2021-01-19 by RootCNC