Duet3D/PanelDueFirmware #96: Encoder input

mikerocklewitz, 2021-04-21:
Ive tried my best to search around the forum and git to get any information.  All i see is one cryptic note about embryonic support being added to one of the 1.x versions....  Whats the status on the encoder input on paneldue?  Thanks so much :)

wilriker, 2021-04-21:
Current status is that software support for the hardware-part of the encoder is there. Also there is one build that will move the X axis at a hard-coded distance and feedrate when the encoder is turned (but that is not published with the releases). There is no encoder support beyond that built in, yet.

mikerocklewitz, 2021-04-22:
Thanks. Ill be patient then. More a want than a need.....would be cool to add to my bluetooth wireless paneldue ;) (Disclaimer: no prints are sent wirelessly)

wilriker, 2021-04-29:
I will close this issue now. If you have more questions please open a topic in the forum.

closed 2021-04-29 by wilriker