Duet3D/PanelDueFirmware #101: Rework Tool Handling

wilriker, 2021-04-30:
## Problem
As of right now there is the problem that screen estate is very limited when we want to control tools/beds/chambers. For beds/chambers the rows we have (button to activate/disable, current temp, target active temp, target standby temp) are ususally sufficient. But for tools we can only cope with very simple tools.

## Proposed Solution
When a tool is determined to be a complex tool (e.g. multiple fans, multiple heaters, multiple extruders, extruder + spindle, anything other out of the ordinary):

Instead of selecting/deselecting a tool when clicking the button of that tool it will instead open a pop-up with detailed controls for this tool. We need to cover the following things.
- A tool can have multiple heaters assigned, so they all need to show up here (we already do this optionally on the main screen)
- A tool can have multiple fans assigned - as of now we only control either the first fan of a tool or fan 0 if no tool is selected
- A tool can be either an extruder, a spindle or a "tool-less" tool, e.g. a sensor head on a tool-changer
- There needs to be a button to select a tool (if not selected) or deselect it
- Control extrusion multipliers of the tool's extruder(s)
- Optional: get access to the tool's offsets (this might open a new pop-up)

## Challenges
- This needs to work for both 4.3" as well as 5"/7" versions - as of right now the smaller resolution can have less displayable elements
- Since the pop-up needs to be reconfigured for every tool the hide-show-order needs to be kept in mind

dc42, 2021-04-30:
Sounds good. I came to the view a year or two ago that we needed to show tool details in a page or popup for just one tool, because there isn't enough space to display several complex tool configurations on a single page.

mfs12, 2021-05-07:
Related to #35.