Duet3D/PanelDueFirmware #102: Make other axes than Z available for M291 movement pop-up

wilriker, 2021-04-30:
## Problem
Currently PanelDueFirmware only supports movement buttons for Z axis for a `M291` command. If the user wants to move any other axis there are no buttons for this purpose.
Screen estate is limited and Z movement was required most of the time so this was an acceptable compromise.

## Proposed Solution
The pop-up needs to be examined for available space to place axis-selector buttons. These will be shown based on what the user as has requested as axis-to-move and tapping one of them will make the available buttons work on that axis instead of being hard-coded to Z.

dc42, 2021-04-30:
Sounds good to me. Has anyone requested this ability to babystep axes other than Z? I guess it could be useful after resurrecting a print, if XY homing isn't precise.

wilriker, 2021-05-07:
I cannot point to a specific user request but the limitation to only jog the Z axis via `M291` was mentioned in multiple occasions and I always thought it would be a good idea to remove limitations in PanelDue firmware as much as possible.

dc42, 2021-08-04:
The axes mentioned could include any of XYZUVWABCD and even more in RRF 3.3 and later on Duet 3 MB6HC. However it is probably sufficient to allow only XYZUVWABCD. We will likely need to enforce and document a limit on how many axes PanelDue can handle in this dialog.

PR #226 (2021-11-07): Add more axis to movement popup
Closes #102.
commit 29819d66 (2021-11-07): Merge pull request #226 from mfs12/add-more-axis-to-movement-popup

Add more axis to movement popup

Closes #102.
closed 2021-11-07 by mfs12