Duet3D/PanelDueFirmware #114: Add network information to status tab

mfs12, 2021-05-21:
I would find it helpful to get more details about network configuration in the status screen not only with a popup.

- ip address
- subnet mask
- gateway
- (hostname if available) not sure if this is possible
- wifi status
- wifi ssid

wilriker, 2021-05-21:
Keep in mind that this information is only available on standalone-boards and won't be available with SBC-connected boards (network-data is only maintained within DCS there and is not available to direct-to-mainboard-connected devices like PanelDue).

wilriker, 2021-05-21:
Also keep in mind that especially on the 4.3" PanelDue there is less space available to display data so it is unlikely that you can get away without a pop-up.

mfs12, 2021-05-31:
@wilriker you added somehting about ip addresses. can this issue then be closed? Can you refer as well to a pull request or commit introducing this changes?

wilriker, 2021-05-31:
Commit 357c51e introduced showing the current IP address (only available for standalone-setups, does not work for SBC setups) on the Setup tab.

If this can be closed depends on whether or not you want to have all the other listed information shown as well since the aforementioned commit only introduced IP address.

mfs12, 2021-06-03:
> Commit 357c51e introduced showing the current IP address (only available for standalone-setups, does not work for SBC setups) on the Setup tab.

in case of SBC setup the address is always 0.0.0.0.