Duet3D/PanelDueFirmware #120: Add support for alternative backlight inverter chip

mfs12, 2021-06-08:
- add support for alternative backlight inverter chip ([MP3202](https://www.monolithicpower.com/en/documentview/productdocument/index/version/2/document_type/Datasheet/lang/en/sku/MP3202/document_id/1495/)) which accepts a PWM frequency of 1KHz maximum.
- 5i/7i uses a pwm frequency of 20kHz
- [datasheet]https://www.monolithicpower.com/en/documentview/productdocument/index/version/2/document_type/Datasheet/lang/en/sku/MP3202/document_id/1495/)

See this code in src/Hardware/Buzzer.cpp:
```c
#if IS_ER
	static const uint32_t backlightPwmFrequency = 20000;	// 20kHz is recommend by East Rising
#else
	static const uint32_t backlightPwmFrequency = 300;		// Working range is about 100Hz to 1KHz. MP3202 dataseet says use 1kHz or below due to soft start. Some frequencies causes flickering on the 4.3" display.
#endif
```

- do a build of the 5i/7i firmware that uses 300Hz the same as the others, and test it.
- make sure that the brightness can be varied and that the **PWM frequency is not audible**.

mfs12, 2021-06-08:
# pwm frequency test

- 100Hz - in dark theme 2 very light flickering at 50% pwm, acceptable buzzing
- 150Hz - in dark theme 2 very light flickering at 50% pwm, acceptable buzzing
- 200Hz - no flickering realizable, acceptable buzzing
- 300Hz - no flickering realizable, acceptable buzzing
- 700Hz - no flickering realizable, pitch gets higher
- 1000Hz - no flickering realizable, pitch gets higher

mfs12, 2021-06-08:
define IS_ER changes colour scheme as well.

mfs12, 2021-06-09:
After doing some more tests yesterday night, probably it's ok to stay with either 200hz or 300Hz. With less than 200Hz you can realize a very light flickering in "Dark theme 2"; although not with others. I prefer the lower pitched sounds. The distance until I can hear the sound is about 15cm.

mfs12, 2021-06-09:
> The manufacturer of the display has offered to add a RC filter so that we can stay with 20kHz. In view of the buzzing at lower PWM frequencies, we will go with this solution. However, we need to reduce the PWM range when the new chip is used, because a range of 0.65V to 1.45V is sufficient to go from off to full brightness. I have asked for one of the unused MCU pins to be grounded (probably PB13) so that we can tell whether the new chip is installed.

mfs12, 2021-06-10:
> Provisional spec for backlight change:

- during init, set pin PB13 to be an input with pullup enabled.
- to determine which backlight inverter is fitted, read from PB13. 
  - This could be done at init time and the result stored, however it is probably easier to do it every time the backlight PWM is to be changed. 
  - LOW indicates new backlight chip, HIGH indicates old chip
- if it is the new chip, scale the PWM appropriately by translating 0 to 100% PWM to the range 20% to 44%, i.e. off is 20% and full brightness is 44%.

mfs12, 2021-06-16:
@dc42 after refactoring the backlight driver turns out the current uses high pwm polarity, does the new one behave the same?

dc42, 2021-06-16:
Yes it does.

PR #127 (2021-06-24): Add backlight pwm mp3302
Closes #120.
closed 2021-06-24 by mfs12