Duet3D/PanelDueFirmware #121: Enable headless builds

mfs12, 2021-06-10:
At the moment the build system depends on eclipse. Which is heavy duty and needs a human to interact with. To get one step closer to continuous integration, automated testing, easy regular nightly builds a headless build system is mandatory.

Internally eclipse uses makefiles but which are auto-generated and are strongly dependent on the host computer.

This could be interesting candidates as replacement:
- makefiles,
- cmake,
- qmake,
- ninja-build,
- meson,
- build root

Or check https://alternativeto.net/software/gnu-make/

dc42, 2021-06-10:
We need to avoid duplicating all the project settings in manually maintained makefiles because of the effort involved, and they would inevitably get out of step. In what way are the generated makefiles strongly dependent on the host computer?

mfs12, 2021-06-10:
> In what way are the generated makefiles strongly dependent on the host computer?

The include directives use absolute paths.

mfs12, 2021-06-10:
> We need to avoid duplicating all the project settings in manually maintained makefiles because of the effort involved, and they would inevitably get out of step.

Yes, this should be definitely a goal.

mfs12, 2021-06-28:
# Howto in cmake
- define toolchain
- define linker scripts
- define custom file conversion (objcopy, other package or release steps)
- define includes
- define configuration
- define linking of libraries

mfs12, 2021-06-28:
https://five-embeddev.com/baremetal/cmake/

mfs12, 2021-08-10:
We decided that RRFlibraries to add CMakeList.txt to the repo so it will support either, Eclipse and cmake.

PR #159 (2021-08-13): Integrate cmake build system
Closes #121.
closed 2021-08-13 by mfs12