Duet3D/PanelDueFirmware #122: Polling interval in normal mode is too slow

mfs12, 2021-06-10:
- normal polling is 0.5s
- pause between response and request 0.7 * 0.5s (PrinterResponseIntverval)
- File or Macro access has higher prio than other requests

There's also a ratelimiting mechanism which slows down polling if requests fail.

wilriker, 2021-06-10:
> File or Macro access has higher prio than other requests
This is only true in case `OkToSend()` returns `true` - otherwise or if no file list fetching was due yet, continue fetching regular status updates.

mfs12, 2021-06-15:
List of status:
|status|OkToSend|whynot|
|---|---|---|
|connecting|false|?|
|idle|true|?|
|printing|true|?|
|stopped|false|?|
|configuring|false|?|
|paused|true|?|
|busy|false|?|
|pausing|false|?|
|resuming|false|?|
|flashing|false|?|
|pausing|false|?|
|toolChange|false|?|
|simulating|false|?|
|off|true|?|

mfs12, 2021-06-15:
> The wanted behavior is simple: status updates as often as possible while maintaining a stable response rate. That's why it starts at a 0.5s interval and decreases this interval slowly when it encounters that RRF was too often unable to actually reply (most likely because PanelDue is contending for resources with a parallel running DWC). Low-prio requests (as file list updates) should be limited to certain statuses as they are already. Status-updates should always be performed. User input needs to be sent when the user presses a button. While screensaver is active, poll-rate can be reduced significantly because updates will not be rendered anyway but we still need status-polls because the response might contain reasons to wake the screen up (apart from the user touching it).

mfs12, 2021-06-17:
Done.

PR #124 (2021-06-17): Fix normal polling interval
Closes #122
commit c6d96170 (2021-06-17): Merge pull request #124 from mfs12/fix-normal-polling-interval

Fix normal polling interval. Closes #122.
closed 2021-06-17 by mfs12