Duet3D/PanelDueFirmware #125: Add Left and right arrows in panel due keyboard.

erwaly, 2021-06-13:
I do not know if it is the right place to request this if it does not correspond delete it or notify me and I will delete it, thanks. Could you add left and right arrow on the keyboard of the panel due? It would be useful when you want to send the same command by changing some value.

mfs12, 2021-06-13:
hey @erwaly , from what i understand you wanna navigate in the entered text so you can modify portions of it?

erwaly, 2021-06-13:
If that possibility would be great especially when you do tests from the panel, it would be necessary to delete and rewrite everything, just change the desired value by navigating through the text produced with a left and right arrow.

mfs12, 2021-06-13:
@erwaly did you see there's already a command history?

which commands do you want to access with this feature?

erwaly, 2021-06-13:
@mfs12 If I saw them, I don't mean that. For example, if in the panel I enter M201 X50 Y50 Z50 and I send I press the up arrow on the keyboard and the last code sent appears again, ok, now if I want to send the same value, changing for example the value from X to X100, I would have to delete the text until leave G1 X and retype everything if the keyboard had a left and right arrow to navigate through the entered text, you would only have to delete the value of X and put the new one. I don't know if you understand me better now?

Picture for reference:

[https://ibb.co/8KKts68](url)

mfs12, 2021-06-13:
thanks for the detailed information. i thought you mean that. just wanted to have a better idea about your use-case. Sometimes building something even more specific could makes sense. I agree the described generic functionality would be handy.

erwaly, 2021-06-13:
If the truth is that it would be useful for example doing printing tests if you want to test different acceleration values, jerk ... it would be great if it could be added, thank you very much for your time!

elpeter25, 2021-06-13:
Totally agree!! I've missed arrows many times for editing entered commands in place.

Cheers!

alienboyxp, 2021-06-13:
Nice improvement!!!

dc42, 2021-06-14:
So what's actually requested is a more general editing function when entering commands, including left/right arrow keys. Would this not also require an Insert key, to toggle between insert and overwrite mode?

mfs12, 2021-06-14:
I think this would be definitely another neat feature to improve the editing experience. although i don't think they depend on each other. I consider the movement keys more important.