Duet3D/PanelDueFirmware #131: Time left on print based on layer completion time is no longer functioning.

MrSteve920, 2021-06-18:
Per the Duet RepRap firmware 3.3 changelog: "Estimated print time provided by the slicer (using M73 commands if present) is now available. Estimated print completion time based on layer progress has been removed."

Because of this change, the paneldue is displaying N/A for the time left on a print based on layer completion on the Print menu.

This should be changed to display the time left on a print based on the slicer reported time remaining instead of the layer completion time. Values reported by slicer via M73 command should be displayed if available, otherwise paneldue should be displaying "slicer N/A" instead of "layer N/A".

[image]

mfs12, 2021-06-18:
@MrSteve920 can you please provide the following information:

- which paneldue hardware are you using?
- which paneldue firmware version are you running?
- which reprap firmware are you running?

MrSteve920, 2021-06-18:
- Paneldue hardware v2. 
- Paneldue firmware 3.2.11. Issue was also present on 3.2.9.
- Reprap firmware 3.3 stable branch

dc42, 2021-06-18:
RRF no longer generates print completion time estimated based on layer completion. See the RRF 3.3 upgrade notes.

mfs12, 2021-06-22:
https://forum.duet3d.com/topic/21735/poll-time-estimates-on-paneldue-5-7/23

MrSteve920, 2021-06-22:
I think we're all in agreement that reprap 3.3 removed the print time estimate based on layer completion. I already indicated that I saw this was removed per the 3.3 upgrade notes.

I'm suggesting that moving forward the paneldue should be updated to no longer display the time left based on layer on reprap 3.3 onward the display will continue to show "layer n/a" which is fairly useless now that this has been removed.

This could be resolved by doing the following per my original comment:
This should be changed to display the time left on a print based on the slicer reported time remaining instead of the layer completion time. Values reported by slicer via M73 command should be displayed if available, otherwise paneldue should be displaying "slicer N/A" instead of "layer N/A".

An alternative option:
If it's not possible to display the time estimated based on slicer provided values then it would at least be cleaner to have the paneldue not display "layer N/A" if reprap firmware >=3.3 is being used.

mfs12, 2021-06-22:
@MrSteve920 With commit fcbc22a4 the layer time was removed. It will be released with the upcoming 3.3 release.

Are you interesting in testing?

MrSteve920, 2021-06-22:
I would rather not be involved with testing at this time. That commit does look like it will resolve this though from what I can tell.

closed 2021-06-22 by MrSteve920