Duet3D/PanelDueFirmware #169: Implement password protected screen locking

mfs12, 2021-08-06:
### Description

The Duet forums have wishlist sections for feature requests if you'd like to check if your request already exists or to discuss your feature request there first.

https://forum.duet3d.com/category/9/hardware-wishlist
https://forum.duet3d.com/category/8/firmware-wishlist
https://forum.duet3d.com/category/11/duet-web-control-wishlist
https://forum.duet3d.com/category/14/paneldue

## Description
Please add a detailed description of the feature.

## What problem does this feature try to solve?
Please add a clear and concise description.

## Proposed Solution
Please add a clear and concise description.

Please add any further or additional information like photos, sketches, mockups, etc.

mfs12, 2021-08-06:
- where to store the password?
  - store it on the printer because finally this what is going to be protected?
- how to store the password? just a checksum of the password with a nuance would be enough.
- how should the dialog look like?
  - on top of the screensaver?
  - what if screensaver is disabled?
- how many fails should be allowed?
- what if the device gets reset?
- howto prevent bruteforce attacks?

dc42, 2021-08-10:
The primary requirement I am aware of is to operate in a mode in which the display works but you can't enter any commands. I will call this Command Locked mode. Here is a possible outline specification:
- Have a facility to set and store a password or PIN. This could be stored in flash memory or PanelDue, or perhaps in config.g and sent to PanelDue.
- Have an option on the Setup page to put the PanelDue into Command Locked mode. PanelDue will switch to the Status screen.
- When in Command Locked mode, the only buttons that can be actioned will be the Control, Status and Console page selection buttons. If any other button is pressed, a PIN or password popup dialog will be displayed.
- When the PIN or password popup is displayed, if you enter the correct PIN or password (probably followed by pressing an OK button) then PanelDue returns to normal mode
- The PIN/password popup will also have a Cancel button, which removes the popup
- If the PIN/password popup is displayed and no button is pressed for some time, the popup is removed. This could be done on the dimming timeout, i.e. if the dimming timeout expires then as well as dimming the screen, the popup is removed.
- It should be possible to put PanelDue into Command Locked mode from RepRapFirmware.
- If we decide to store the PIN or password in PanelDue flash memory, then it must be possible to reset or clear the PIN/password from RRF, to provide recovery from a lost PIN or password.
- While in Command Locked mode, popup messages continue to be displayed as normal, however any attempt to acknowledge or cancel a message that requires a response will invoke the PIN/password popup.

mfs12, 2021-08-12:
@dc42 

> Have a facility to set and store a password or PIN. This could be stored in flash memory or PanelDue, or perhaps in config.g and sent to PanelDue.

RRF already has a password. Would it make sense to re-use that? then there would be only one password for the system. And the recovery can be done completely with existing infrastructure. Checking with the gcodes, i realized there's not yet a command to verify the password but only to set it.

mfs12, 2021-08-12:
I think either everything is configured from RRF, i.e. password, command locking / normal mode, timeout, which is provided via OM or everything is handled by PD. I think only the password might be handled still by RRF so there's always a "simple" way of recovery using an already established workflow.

dc42, 2021-08-12:
The present RRF password scheme is insecure in that the password appears in plain text in config.g. Although we are not looking for high security, I think we should do better for this feature. One possibility is that we add a command to set the PanelDue password (e.g. M551 with a S parameter to ay which password is being set), then we send a salted hash of that password to PanelDue, which stores it in NVM if it is different from the one already stored. That means users do not have to put that command in config.g.

mfs12, 2021-11-16:
Related to #71.

dc42, 2025-05-09:
We don't plan to do this. PanelDue is being superseded by a new screen. I will continue to implement changes where they are needed to retain compatibility with new RepRapFirmware versions for some time, but it is unlikely that we will implement new features.

closed 2025-05-09 by dc42