Duet3D/PanelDueFirmware #244: Add option to sort files by data in file list

dc42, 2022-01-30:
### Description

It would be helpful to be able to sort the list of files in a folder pn the SD card by timestamp as an alternative to being sorted by name. See https://forum.duet3d.com/topic/27060/m20-option-to-sort-by-date.

Note: to implement this, PanelDue will need to store the time stamp of each file (possibly in compressed format) and RRF will need a new M20 option to return the time stamp along with the filename.

goscicki, 2022-02-04:
I would love to see this feature implemented as well. It is a pain finding a file that was just uploaded to a folder with hundreds of files.

JannesV, 2022-07-22:
I'm going to give this a bump since I'd love to see this feature too!

MarkoP2507, 2024-10-19:
This feature would simplify many things

dc42, 2025-05-11:
We don't plan to do this. PanelDue is being superseded by a new screen using more capable hardware. I will continue to implement changes where they are needed to retain compatibility with new RepRapFirmware versions for some time, but it is unlikely that we will implement any new features.

closed 2025-05-11 by dc42