Duet3D/PanelDueFirmware #245: Multiple bed/chamber heaters temperature sets only the 1st

OasisOfChaos, 2022-01-30:
### Description

## Description
If you define more than one bed/chamber heater and set the active/standby temperature on the 2nd, 3rd or 4th, the temperature of the first is modified. The changes made via the web interface are displayed as they should.

PanelDue always sends M14x P0 Sx or M14x P0 Rx (I verified this).

## Versions
3.4.0

## Process
Define more than one of the heaters above, with M140 H4 P1 or M141 H4 P1 and modify the active temperature on the panel. The one defined with P0 will change.

mfs12, 2023-04-05:
hey @OasisOfChaos are you still there. I would fix it, could you share a configuration with me to reproduce the bug?

mfs12, 2023-04-12:
@OasisOfChaos does this problem still apply with version paneldue-3.4.1?

mfs12, 2023-04-13:
please provide a configuration to reproduce this bug.

mfs12, 2023-04-14:
```
; GENERAL
G90                                                       ; send absolute coordinates...
M83                                                       ; ...but relative extruder moves
M550 P"ARIES"                                             ; set printer name

; NETWORK
M552 P0.0.0.0 S1                                          ; enable network and acquire dynamic address via DHCP
M586 P0 S1                                                ; enable HTTP
M586 P1 S0                                                ; disable FTP
M586 P2 S0                                                ; disable Telnet

; PANEL DUE SERIAL PORT
M575 P1 S1 B57600

; DRIVES
M569 P0 S0                                                ; physical drive 0 goes forwards
M569 P1 S1                                                ; physical drive 1 goes forwards
M569 P2 S1                                                ; physical drive 2 goes forwards
M569 P3 S1                                                ; physical drive 3 goes forwards
M569 P4 S1                                                ; physical drive 4 goes forwards

M584 X0 Y1 Z2:4 E3                                        ; set drive mapping - DUAL Z

M350 X16 Y16 Z16:16 E16 I1                                ; configure microstepping with interpolation
M92 X80.00 Y80.00 Z1600.00 E415.00                        ; set steps per mm
M566 X400.00 Y400.00 Z0 E120.00                           ; set maximum instantaneous speed changes (mm/min) JERK
M203 X12000.00 Y12000.00 Z900.00 E1200.00                 ; set maximum speeds (mm/min)
M201 X800.00 Y400.00 Z20.00 E250.00                       ; set accelerations (mm/s^2)
M906 X1400 Y1400 Z1200 E800 I50                           ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                                   ; Set idle timeout

; DUAL Z
M671 X20:460 Y250:250 S5.0   							  ; leadscrews at left (connected to Z) and right (connected to E1) of X axis

; AXIS LIMITS
M208 X0:504 Y0:500 Z-1:400 

; ENDSTOPS
M574 X1 S1 P"!^xstop"									  ; configure active-high endstop for low end on X via pin xstop
M574 Y1 S1 P"!^ystop"                                     ; configure active-high endstop for low end on Y via pin ystop
M574 Z1 S2                                                ; configure Z-probe endstop for low end on Z

; Z-PROBE
M574 Z0 C"nil"                                   		  ; no Z endstop switch, free up Z endstop input
;M558 P8 C"!zprobe.in" H5 F1200 T12000 R0.5 A4 S0.05      ; PIEZO - P8 unfiltered, P5 filtered, P8 is faster
M558 P9 C"^zprobe.in" H5 F300 T10000 A10 S0.03 R0.3		  ; BLTOUCH
M950 S0 C"exp.heater3"
;G31 P100 X0 Y0 Z-0.09									  ; PIEZO OFFSET
G31 P100 X-22 Y0 Z2.690						    	      ; BLTOUCH OFFSET
M557 X30:480 Y30:480 S75                                  ; define mesh grid

; BED - ZONE 01 HEATER (CENTER)
M308 S0 P"bedtemp" Y"thermistor" T100000 B3950            ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bedheat" T0                                     ; create bed heater output on bedheat and map it to sensor 0
M307 H0 B1 S1.00 A97.3 C230.7 D2.8 S1.00 V24.2            ; enable bang-bang mode for the bed heater and set PWM limit
M140 H0 P0                                                ; map heated bed to heater 0
M143 H0 S120                                              ; set temperature limit for heater 0 to 120C
M570 H0 S200                                              ; Set maximum heating time (Snnn, in seconds).

; BED - ZONE 02 HEATER (EDGE)
M308 S2 P"e1temp" Y"thermistor" T100000 B3950             ; configure sensor 2 as thermistor on pin e1temp
M950 H2 C"e1heat" T2                                      ; create bed heater output on bedheat and map it to sensor 0
M307 H2 B1 S1.00 A166.8 C910.6 D2.8 V24.2                 ; enable bang-bang mode for the bed heater and set PWM limit
M140 H2 P1                                                ; map heated bed to heater 1
M143 H2 S120                                              ; set temperature limit for heater 0 to 120C
M570 H2 S200                                              ; Set maximum heating time (Snnn, in seconds).

; MOSQUITO MAGNUM HEATER & THERMISTOR
M308 S1 P"e0temp" Y"thermistor" T500000 B4723 C1.19622e-7 ; configure sensor 1 as thermistor on pin e0temp
M950 H1 C"e0heat" T1                                      ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 R2.531 K0.649:0.000 D4.51 E1.35 S1.00 B0 V24.1    ; disable bang-bang mode for heater and set PWM limit

; CHAMBER THERMISTOR
M308 S3 P"exp.thermistor3" Y"thermistor" T100000 B9350 A"CHAMBER" 

; FILAMENT SENSORS
;M591 D0 P7 C"^e0stop" L7 R50:250 E22 S1    			  ; pulse, disabled, 7 mm/pulse, measure every 22 sec
M591 D0 P5 C"e0_stop" R40:1200 E3.0 S0                    ; DUET laser filament sensor

; FANS
M950 F0 C"fan0" Q500                                      ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                            ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"fan1" Q500                                      ; create fan 1 on pin fan1 and set its frequency
M106 P1 S0 H-1                                            ; set fan 1 value. Thermostatic control is turned off

; TOOLS
M563 P0 D0 H1 F0                                          ; define tool 0
G10 P0 X0 Y0 Z0                                           ; set tool 0 axis offsets
G10 P0 R0 S0                                              ; set initial tool 0 active and standby temperatures to 0C

; ACCELEROMETER
M955 P0 C"spi.cs4+spi.cs3"
M593 P"zvddd" F52.2 ; use ZVD input shaping to cancel ringing at 37.6Hz

; MISCELLANIOUS
T0                                                        ; select first tool
M572 D0 S0.07                                             ; PRESSURE ADVANCE
M98 P"startup.g"
```

closed 2023-04-14 by mfs12