Duet3D/PanelDueFirmware #268: Tool data is overlaid on top of macro filenames

dc42, 2022-06-28:
### Description

On the Control page, if the machine has a large number of tools then the temperatures for the higher tool numbers overwrite the list of available macros at the right hand side of the screen. This has two consequences:
1. Some parts of some letters from the macro filenames show between the tool temperature lines, which looks messy.
2. When trying to touch the active or standby temperature field to change it, sometimes the macro is triggered instead.
The old code used to display the macro file list only if the number of tools displayed was low enough to leave room for the macro file names. This behaviour should be reinstated.

mfs12, 2023-04-05:
@dc42 can you please provide a configuration to reproduce this bug?

mfs12, 2023-04-13:
@dc42 can't reproduce this issue.

/close

closed 2023-04-13 by mfs12