Duet3D/PanelDueFirmware #274: Vertical layout

MilanBacaCz, 2022-07-30:
### Description

Have a vertical layout on panel due would be nice.

dc42, 2025-05-09:
We don't plan to do this. PanelDue is being superseded by a new screen. I will continue to implement changes where they are needed to retain compatibility with new RepRapFirmware versions for some time, but it is unlikely that we will implement new features.

closed 2025-05-09 by dc42

kevenwyld, 2025-05-10:
> PanelDue is being superseded by a new screen.

@dc42 is there a way to follow development for that somewhere?  (Sorry for the off topic reply, just not sure where else to ask)

dc42, 2025-05-11:
The repo for that development isn't public yet. We're currently testing prototypes of the new screen.