Duet3D/PanelDueFirmware #279: Paneldue unneccicary info messages

Modix3D, 2022-11-21:
### Description

## Description
Very frequent info: status changed messages. These don't add any useful information, but can clog the PanelDue config

https://forum.duet3d.com/topic/29951/paneldue-getting-frequent-info-popups-clogging-the-console

## Versions
Paneldue version 3.4.1, ReprapFirmware 3.4.2

## Process
Happens on any bootup or use of the printer. Especially if there is a less than ideal WiFi setup

## Experienced Result
Very frequent info: status changed messages. These don't add any useful information, but can clog the PanelDue config
[image]

## Expected Result
Info: status messages being either disabled, or able to set them using a variable. Useful for debugging, but not for day-to-day use.

dc42, 2023-11-01:
I suggest we generalise this to a switch for Normal/Verbose status messages, because in future we may wish to add additional message in verbose mode. The default shood be Normal.

mfs12, 2023-11-02:
Added toggle switch to setup up page. Setting is stored in nvram.

PR #320 (2023-11-02): MessageLog: introduced loglevel normal and verbose
Closes #279.
closed 2023-11-02 by mfs12