Duet3D/PanelDueFirmware #283: Fix qoi thumbnail selection

mfs12, 2023-04-04:
### Description

## Description
Thumbnail selection doesn't work in certain situations.

https://forum.duet3d.com/topic/31919/no-qoi-thumbnail-displayed-superslicer-2-5-59

## Versions
- PD 3.4.1

## Process
- refer to forum

## Experienced Result
no thumbnail is displayed

## Expected Result
thumbnail is displayed

dc42, 2023-04-04:
This may be a RRF issue, caused by the first thumbnail being so long that RRF doesn't look beyond it. I am running some tests.

dc42, 2023-04-04:
Confirmed, this was a RRF issue. On my test file with 300x300 followed by 160x160 thumbnails, RRF only reported the first one. I have increased the amount of the file we scan. Now the second one is also reported and PanelDue displays it.

closed 2023-04-04 by dc42

mfs12, 2023-04-05:
@dc42 did you test it with the files from marveneer?

dc42, 2023-04-05:
No, I reproduced it with a file I generated form PrusaSlicer, and tested the fix on the same file.