Duet3D/PanelDueFirmware #285: Implement support for new M291 message types

mfs12, 2023-04-04:
### Description

## Description

## What problem does this feature try to solve?
Please add a clear and concise description.

## Proposed Solution
Please add a clear and concise description.

Please add any further or additional information like photos, sketches, mockups, etc.

mfs12, 2023-04-05:
```cpp
{ "axisControls", OBJECT_MODEL_FUNC_IF(self->mode == 2 || self->mode == 3, (int32_t)self->controls.GetRaw()),	ObjectModelEntryFlags::important },
{ "cancelButton", OBJECT_MODEL_FUNC(self->limits.canCancel), ObjectModelEntryFlags::important },
{ "choices", OBJECT_MODEL_FUNC_IF(self->mode == 4, self->limits.choices),	ObjectModelEntryFlags::important },
{ "default", OBJECT_MODEL_FUNC_IF(self->mode >= 4, self->limits.defaultVal),	ObjectModelEntryFlags::important },
{ "max", OBJECT_MODEL_FUNC_IF(self->mode >= 5, self->limits.maxVal), ObjectModelEntryFlags::important },
{ "message", OBJECT_MODEL_FUNC(self->message.c_str()), ObjectModelEntryFlags::important },
{ "min", OBJECT_MODEL_FUNC_IF(self->mode >= 5, self->limits.minVal), ObjectModelEntryFlags::important },
{ "mode", OBJECT_MODEL_FUNC((int32_t)self->mode), ObjectModelEntryFlags::important },
{ "seq", OBJECT_MODEL_FUNC((int32_t)self->seq), ObjectModelEntryFlags::important },
{ "timeout", OBJECT_MODEL_FUNC((int32_t)self->timeout), ObjectModelEntryFlags::important },
{ "title", OBJECT_MODEL_FUNC(self->title.c_str()), ObjectModelEntryFlags::important }
```

mfs12, 2023-04-06:
@dc42 is there a max length of strings to enter? if so what is it?

dc42, 2023-04-06:
The only limits currently are:
1. Maximum length of any GCode command, including M291 commands. This is currently 256 bytes. However, a command that uses variable substitution could exceed this limit.
2. Currently, indiividual strings can't exceed 100 characters each. However, this is about to be increased to 256.
I suggest you define and we document a maximum length of each string and a maximum number of choices that can be displayed on PanelDue. This number could be lower on the 4.3" display than on the 5" and 7". Longer strings should be truncated, which the text writing functions on PanelDue do anyway.

dc42, 2023-04-06:
That sounds reasonable. What length text will fit on the 800x480 screen of the 5" and 7" displays?